mhendriks/P1-Dongel-ESP32 #4: Last and current hour data not shown in Hours tab

ttonino, 2022-12-28:
The "hours" usage table does not show the consumption of the current hour, nor of the hour before.
Example: it is now 9:23 local time.
The Hours table starts with 07:00-07:59, under it, the data for 06:00-06:59 is shown.
Thus, I cannot see what has been consumed from 08:00-08:59.

Time code from Actueel page is 221228092540W. Time displayed on top right of Hours page is 2022-12-28 09:25:46. First hour displayed in Hours table is (28) [07:00 - 07:59].

This has started the moment I upgraded to 4.6.0. On 4.5.0, the previous hour would be shown correctly.

(I think that the current hour was displayed before I upgraded to 4.5.0. - but I'm not sure. The info for the current day is displayed and updated in the Days tab in 4.6.0.)

mhendriks, 2022-12-29:
Did you look at the bottom of the table? some times the new values are added at the bottom.

See below ... just happend.

[image]

ttonino, 2022-12-29:
The data of the previous hour is now on top of the table. It is 10:30 and the (29) [09:00 - 09:59] entry is at the top. So I cannot confirm it was at the bottom, sorry.

ttonino, 2022-12-29:
Indeed, it seems the entries are added at the bottom sometimes. In this case, multiple entires.
[image]

mhendriks, 2023-01-05:
in the 4.6.2 release the table refresh is fixed. The 4.6.2 version will be released very soon.
The frontend part (with the fix) will be fetched automatically when the dongle is on firmware 4.6.x and the indexfile is pointing to @4.6.

closed 2023-01-05 by mhendriks