mhendriks/P1-Dongel-ESP32 #5: p1 pro does not reconnect after wifi failure

githubcdr, 2023-01-04:
Hi,

I noticed that my p1-pro meter stays disconnected when my wifi network is unavailable for a while.

It would be nice if a watchdog process would trigger on this event and restart the connection or the device. For now I have to manually restart the device to get it back online.

mhendriks, 2023-01-07:
When the WiFi fails, the Pro dongle will try to reconnect endlessly (once every 10 seconds). Just did a test with the network off.

[image]

SteGe88, 2023-01-10:
Same happens to me (just happend, when rebooted my AP, the Pro didn't came back online, nor was it pushing messages to MQTT).

mhendriks, 2023-01-10:
Did the WiFi device turn off completely or, for example, only a disable of the 2.4G channel?

SteGe88, 2023-01-10:
Both cases result in the same issue I think :-) I have noticed it before when my WiFi controller automatically search for the best channel and strength, the connection with the Pro has been broken. Today I fully rebooted both the access points and the Pro didn't came back online afterwards (before it was working). It felt like the problem with the changed channel was resolved after the last release I had put on the pro, but today I saw it at the reboot of the AP.

githubcdr, 2023-01-10:
Exactly the same issue here, would be great to have a keepalive funtion checking for this issue, it happens quite frequent

mhendriks, 2023-01-12:
The Dongle has an event mechanism that registers disconnects and tries to reconnect once every 5 seconds. That's why I'm looking for more information to reproduce the situation and more importantly .... solve it

SteGe88, 2023-01-12:
How can we help to give you more info Martijn?

mhendriks, 2023-01-12:
I will first do some tests myself to see how the dongle reacts in different situations:
- wifi on/off/on (also for few minutes)
- router reboot
- Wi-Fi signal interference

mhendriks, 2023-01-13:
I think I know in which situation a deadlock occurs. Tip from another user. In the event of a power failure, the dongle will start up faster than the DHCP server. The dongle does not receive an IP address and will not request it again. Now investigate a method to keep requesting the IP address from the DHCP server or else disconnect and restart.

ikbenfrank, 2023-05-08:
I think I have a similar problem. My situation:
- Asus router which has the DHCP server.
- Omada AP's that run Wifi optimization from 2.00 - 2.22am daily.
- P1 dongle pro with manually assigned ip address in the dhcp server.

DHCP request from the P1 dongle between 2 and 3am which gets acknowledged but the dongle doesn't get online after the wifi optimization is done. It always has an end time at 2.21 am. Uptime of the P1 dongle is between 1 and 3 days.
I had firmware 4.7.0 and now 4.8.2 but problem persists.

Router log:
May  8 02:14:30 dnsmasq-dhcp[29832]: DHCPREQUEST(br0) 192.168.2.31 dc:54:75:aa:c7:8c 
May  8 02:14:30 dnsmasq-dhcp[29832]: DHCPACK(br0) 192.168.2.31 dc:54:75:aa:c7:8c p1_dongle_pro

I have disabled wifi optimization for now to see if this fixes it.
If you want me to try something or send a log, let me know.

ikbenfrank, 2023-05-11:
@mhendriks In the search for finding a solution I already tried different power adapters the last few weeks without results. By disabling wifi optimization the p1 dongle hasn't been down yet so it solves the problem for now.

closed 2023-06-22 by mhendriks