mhendriks/P1-Dongel-ESP32 #6: Problems with flashing FW manually

frahlg, 2023-02-04:
Thank you Martijn for your great work,

Sorry to say I’ve had no success with my dongle (hardware works just fine). I’ve installed esphome and my meter unfortunately seems to be an Aidon meter which I think is not sending over serial using DSMR. Just a note, it **currently works now with the firmware AMS Reader** https://github.com/UtilitechAS/amsreader-firmware with an esp8266-device, but I have no success on compiling that FW for the ESP32C3 with the RISC CPU. And I have also not gotten your FW to work at all.

I’ve tried installing https://docs.smart-stuff.nl/geavanceerd/flashen (http://ota.smart-stuff.nl/v5/4-sketch-latest_4Mb.bin ) with esptool.py, BUT it just ends up in a endless boot loop. (a note, I have been successfully flashing esphome and it works, so it's not HW or drivers etc). Could you please guide me here?

mhendriks, 2023-02-04:
H Fredrik,

You're welcome.

I took a quick look at the AMS solution and it doesn't work with (D)SMR as
far as I can tell :-(

The AMS reader should also work on the ESP32 and therefore also on the
ESP32 C3. Only the hardware is not suitable for the AMS meters.

flashing the esp32:
With an ESP32, more components need to be flashed than with an ESP8266.
Below is the command I use to flash with esptool

esptool_py/4.2.1/esptool \
--chip esp32c3 \
--port /dev/cu.usbmodem2101 \
--baud 3000000 \
--before default_reset --after hard_reset write_flash -z \
--flash_mode dio \
--flash_freq 80m \
--flash_size 4MB \
0x0 P1-Installer.ino.bootloader.bin \
0x8000 P1-Installer.ino.partitions.bin \
0xe000 boot_app0.bin \
0x10000 DSMR-API-V4.6.2_4Mb.bin

Att all the bin files.

Does this give some direction ?

Warm regards / Met warme groet,

Martijn | smart-stuff.nl

frahlg, 2023-02-05:
Where do I find the files? I have found `DSMR-API-V4.6.2_4Mb.bin` from your gitbook documentation. Maybe this could be added there in case someone else wants to flash from scratch.

mhendriks, 2023-02-06:
Hi Fredrik,

All files are stored in the http://ota.smart-stuff.nl/v5/ folder.
The folder isn't accessible only to files.

structure is: http://ota.smart-stuff.nl/v5/DSMR-API-V<version>_4Mb.bin

the latest on this moment is :
http://ota.smart-stuff.nl/v5/DSMR-API-V4.7.0SE_4Mb.bin

The SE suffix is needed for the Swedish version (Obis totals instead of
day/night totals).

Thanks for your tip!

Warm regards / Met warme groet,

Martijn | smart-stuff.nl

githubcdr, 2023-04-30:
Maybe make the folder listing available?

mhendriks, 2023-05-01:
Hi Frederik,

Unfortunately, not done on purpose;-)
The goal is to use the update mechanism in the info screen (red text).
[image: Screenshot 2023-05-01 at 22.47.54.png]

Or to use the remote update url with the version number of choice.
for example:

http://p1-dongle-pro.local/remote-update?version=4.8.2

Other new feature is the ESP webtools site (for flashing via usb micro ) :
https://mhendriks.github.io/P1-Dongel-ESP32/

Warm regards / Met warme groet,

Martijn | smart-stuff.nl

closed 2023-06-22 by mhendriks