mhendriks/P1-Dongel-ESP32 #12: Na update p1 dongel niet meer bereikbaar

Habbie1, 2023-02-12:
Heb versie esp32-4.7.3 erop gezet..maar nu  niet meer bereikbaar via WiFi...zie geen ander onbekend adres in mijn netwerk...wat zijn mogelijke oplossingen?

mhendriks, 2023-02-12:
er is gee 4.7.3 firmware beschikbaar. Of bedoelt u de DMSRindexEDGE.html  vervangen? Laatste firmware is 4.7.0.

Habbie1, 2023-02-12:
Had deze versie erop gezet..was dat niet de juiste?

Verzonden vanaf Outlook voor Android<https://aka.ms/AAb9ysg>

mhendriks, 2023-02-12:
Dit is de laatste DSMR-API versie voor de Pro Dongle: https://ota.smart-stuff.nl/v5/DSMR-API-V4.7.0_4Mb

Enkele vragen/tips:
- Brande de blauwe LED niet meer constant? 
- als de dongle even van het stroom afgehaald wordt (5 tellen) en daarna weer ingeplugd ... brand dan kort de blauwe led? knippert deze om de enkele seconde, brand de led constant of helemaal niets ?

Habbie1, 2023-02-12:
Stekker rj11 erin brand  1 maal kort het blauwe ledje daarna niets meer...
Hans

Verzonden vanaf Outlook voor Android<https://aka.ms/AAb9ysg>

mhendriks, 2023-02-12:
Als de dongle niet meer te vinden is als hotspot (duurt soms 45 - 75
seconden na opnieuw starten) dan ben ik bang dat er iets is misgegaan met
updaten.
Eventueel kunt u nog proberen of het met usb adapter aangesloten wel
normaal functioneert. Er zijn drie opties, nl:
1) dongle is het oude netwerk kwijt na update; dan via hotspot opnieuw
aanmelden
2) Bij het opnieuw starten vraagt de dongle toch te veel energie van de
slimme meter waardoor deze wordt afgeknepen... kan is erg klein
3) iets mis gegaan bij het updaten waardoor de dongle niet meer koppelt aan
uw wifinetwerk en ook niet meer vindbaar is als hotspot

Indien het optie 3 is kan de dongle alleen weer tot leven geroepen worden
door deze via de usb kabel opnieuw te flashen. Dit kunt u zelf doen maar u
kunt de dongle ook opsturen naar mijn dan maak ik hem weer in orde.

Habbie1, 2023-02-13:
Nee hij blijft dood...waar kan ik hem na toe sturen??

Heb hier geen mogelijkheid om te flashen

Hans

Verzonden vanaf Outlook voor Android<https://aka.ms/AAb9ysg>

closed 2023-06-22 by mhendriks