mhendriks/P1-Dongel-ESP32 #18: Feature request: Enkel MQTT topic met alle data van api/v2/sm/actual

itavero, 2023-05-08:
Allereerst nogmaals bedankt voor het maken van deze dongle. Ik heb 'm nu ongeveer een week aangesloten en het heeft mij al wat meer inzicht gegeven in ons energieverbruik en de elektrische installatie van onze woning.

Hoewel het dashboard van de dongle zelf best veel biedt, mis ik wel een paar zaken in de logging.
Daarnaast worden volgende maand ook zonnepanelen op het dak geplaatst en ik zou uiteraard graag de data uit de P1 poort combineren met de data van het Enphase systeem.
Op die manier kan ik een dashboard maken (bijvoorbeeld door de data in InfluxDB te stoppen en uit te lezen met Grafana) dat het "echte" energieverbruik laat zien en met de combinatie van deze data kan ik ook andere apparaten gaan automatiseren (voor wanneer het salderen weg gaat en ik zelf mijn eigen energie direct zou willen gebruiken).

Momenteel lijkt de dongle alle informatie die hij heeft wel te publiceren, maar alles op losse topics.
Ik wil graag alle data uit het pakket in een keer ontvangen, zodat ik zeker weet dat de data die ik heb bij elkaar hoort.
Het zou voor mij dus handig zijn als alle data die nu op het `/api/v2/sm/actual` endpoint wordt aangeboden, in hetzelfde formaat op een MQTT topic gepubliceerd wordt.
De andere, individuele topics heb ik in dat scenario niet meer nodig.

Dit bericht zou bij voorkeur gepubliceerd worden wanneer een nieuw telegram ontvangen en verwerkt is uit de P1 poort.
Is dit iets dat toegevoegd kan worden?

PS: Misschien kijk ik er overheen, maar ik mis eigenlijk ook een instelling om in te stellen hoe vaak de meter wordt uitgelezen.

mhendriks, 2023-05-11:
Leuk om te lezen :-)

De gedachte achter de dongle is dat deze data uit de slimme meter verwerkt en inzichtelijk maakt in een eigen website. Daarnaast meerdere manieren om deze data te kunnen delen. 

Enphase heeft een interface / api waar gegevens uit gelezen kunnen worden. Als deze samen met de p1 gegevens in de InfluxDB gestopt worden kunnen er mooie doorsnijdingen gemaakt worden. Er zijn zelfs gebruikers die gegevens combineren in een eigen console oplossing zie: [blog energy dashboards](https://smart-stuff.nl/inspirerende-energie-dashboards/ )

In de DSMRsettings.json kan een optie worden aangezet waardoor alle waardes als 1 json bericht wordt verzonden :  "act-json-mqtt":true . Denk dat dit de optie is:-)

In de settings kan ook het mqtt interval worden ingesteld.  Mqtt bericht wordt verzonden na een succesvolle leesactie en indien het interval verlopen is. snelste optie is 1 seconde.

De dongle past zich aan aan de slimme meter. SMR 5 meters leveren eens per seconde een telegram ouder eens per 10 seconden. Dit is in de nieuwe versies niet in te stellen. Wel is het mqtt verzend interval in te stellen op bv 10 seconde. Dit kan direct in de DSMRsettings.json of via de Instellingen pagina van de dongle.

itavero, 2023-05-11:
Die optie lijkt de juiste. Alle meterdata van de stroom worden dan op 1 topic gezet. Gek genoeg zijn de timestamp en de meterstand van het gas nog wel losse topics.

mhendriks, 2023-05-12:
Top, dat is gelukt. Gas (en ook water) zijn nog geen onderdeel. Dit zal in een van de komende releases meegenomen worden.

closed 2023-10-11 by mhendriks