mhendriks/P1-Dongel-ESP32 #19: Eth-P1-Dongle-Pro na flashen niet meer via Ethernet bereikbaar

blolp, 2023-05-16:
Afgezien van wat dag / maand historie issues werkende eth-dongle-pro met data lokaal. Geen koppelingen nodig of ingesteld. Aanwezige versie 4.8.0-1697 via de interne webinterface getracht te updaten. Daarna geen verbinding meer te krijgen met de dongel. Via smart-stuff de recenste .bin opgehaald via http://ota.smart-stuff.nl/v5/4-sketch-latest_4Mb.bin
Geflasht via de webtool - melding succesvol - en vervolgens aangesloten op P1 en Ethernet. Ethernet verbinding komt niet tot stand, de router toont geen dongle (Connect led op de connector brand wel, Data-led knippert af en toe). Wifi wifispot komt wel op en kan ingevuld. De wens is echter om via de kabel te verbinden ipv Wifi.
Is er een aparte bin-file nodig voor het updaten via micro usb? Zo ja welke?

mhendriks, 2023-05-17:
Hi Leo,

De ethernet versie haalt de binary van een andere plek af, namelijk: http://ota.smart-stuff.nl/eth/. 
voor versie 4.8.0 is dat http://ota.smart-stuff.nl/eth/DSMR-API-V4.8.0_4Mb.bin 

De v5 versie is de wifi versie. Doet het ook op de ethernet dongle maar alleen via wifi. Maar dat had je al gezien ;-)
Advies is om de 4.8.0. (ETH) versie er op te zetten dan werkt de dongle weer zoals het hoort.

De Ethernet versie kan nu ook geflasht worden via de webtool: https://mhendriks.github.io/P1-Dongel-ESP32/

blolp, 2023-05-17:
Flashen lukte perfect en in één keer via de webtool. Dongel aangesloten op smart meter en router via kabel - werkt direct en verzameld data. Bedankt voor het snelle herstel en support!

closed 2023-05-17 by mhendriks