mhendriks/P1-Dongel-ESP32 #20: Foutmelding in home assistant

tjveldhuizen, 2023-06-10:
De MQTT-koppeling geeft in Home Assistant de volgende foutmelding. Eerlijk heb ik geen idee of dat sinds een bepaald moment is, of dat het altijd al zo geweest is.
```
2023-06-10 08:40:33.269 ERROR (MainThread) [homeassistant.components.mqtt.models] Exception raised when updating state of sensor.dsmr_last_update, topic: 'P1-Dongle-Pro/timestamp' with payload: b'230610084029S'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 581, in state
    numerical_value = int(value)
                      ^^^^^^^^^^
ValueError: invalid literal for int() with base 10: '2023-06-10 08:40:29'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/mqtt/models.py", line 270, in process_write_state_requests
    entity.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 590, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 654, in _async_write_ha_state
    state = self._stringify_state(available)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 596, in _stringify_state
    if (state := self.state) is None:
                 ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 585, in state
    raise ValueError(
ValueError: Sensor sensor.dsmr_last_update has device class 'None', state class 'measurement' unit '' and suggested precision 'None' thus indicating it has a numeric value; however, it has the non-numeric value: '2023-06-1ss 'str'>)
```

Ik heb zo'n vermoeden dat in ieder geval de tweede paramater op onderstaande regel (net als de eerste) timestamp moet zijn, gebaseerd op [deze documentatie](https://developers.home-assistant.io/docs/core/entity/sensor?_highlight=total_increasing#long-term-statistics:~:text=SensorDeviceClass.TIMESTAMP). Maar ik snap geloof ik niet zo goed of het formaat van het timestamp dan het juiste is.
https://github.com/mhendriks/P1-Dongel-ESP32/blob/42cb22d8b8bb9c1794542387489aad4352183dd1/MQTT.ino#L42

mhendriks, 2023-06-11:
Ha heeft moeite met de timestamp zo te lezen.
Wat is de HA versie en de DSMR-API versie van de dongle?

tjveldhuizen, 2023-06-11:
HA-versie is 2023.6.0, en de software op de dongle 4.8.3+1817 (17/05/2023)

mhendriks, 2023-06-11:
Ben je goed up to date ;-)

update HA naar 23.6.1 en krijg de fout nu ook te zien ... ga er mee aan de slag

[image]

mhendriks, 2023-06-14:
In de 4.8.4 versie is dit probleem opgelost. Een offset (zone) dient namelijk ook toegevoegd te worden aan de tijd.
4.8.4 Beta kan worden geinstalleerd via http://p1-dongle-pro.local/remote-update?version=4.8.4

closed 2023-06-14 by mhendriks