mhendriks/P1-Dongel-ESP32 #24: MQTT all topic not compatible with Homeassistant auto config

jeroenrnl, 2023-10-14:
When changing the setting `act-json-mqtt` to `true` (which I need to process the data with my home-built energy monitoring system), the autoconfigure data sent to home assistant is still describing the other situation, causing Home Assitant to show all values as 'unknown'.

Sending the `/all` configuration data to Home Assistant might be a little complicated, but maybe it's possible to send both?

mhendriks, 2023-10-14:
I know ... in on purpose ... the /all is added on request.
The Auto Detect (AD) function for HA should be enabled when using /all but the /all json isn't mentioned in the AD structure.

jeroenrnl, 2023-10-15:
I'll pull the source code tomorrow and see if I can enable both at the same time

jeroenrnl, 2023-10-16:
I noticed that with 4.8.6 the gas information is sent with both the single topic and the `/all` topic

jeroenrnl, 2023-10-18:
I have modified the source code to deliver to both topics when `bActJsonMQTT` and `EnableHAdiscovery` are both set, however, I've got troubles compiling the source code, it fails on 
```
DSMRloggerAPI.h:98:32: error: 'peak_pwr_last_q' was not declared in this scope
```
I figured out this has to do with the Belgian and Dutch DSMR system and installed both [dsmr2Lib](https://github.com/mrWheel/dsmr2Lib)  and [arduino-dsmr-be](https://github.com/mrWheel/arduino-dsmr-be) but that didn't solve it.

Oh, and something that cost me a bit of time to figure out is:
```
Arduino-IDE settings for P1 Dongle hardware ESP32:
  - Board: "ESP32 Dev Module"
```
which should be
```
Arduino-IDE settings for P1 Dongle hardware ESP32:
  - Board: "ESP32C3 Dev Module"
```
I could make a pull request for that.

jeroenrnl, 2023-10-18:
Yes, I have changed to "ESP32C3 Dev Module", I meant that the code still says "ESP32 Dev Module" and it took me a bit of time to figure out.

I have already installed DSMR2Lib, but I still get the error
```
DSMRloggerAPI.h:98:32: error: 'peak_pwr_last_q' was not declared in this scope
```
and `peak_pwr_last_q` doesn't appear in `DSMR2Lib`, because `peak_pwr_last_q` seems to be related to the Belgian DSMR variant, I tried adding [arduino-dsmr-be](https://github.com/mrWheel/arduino-dsmr-be), but that didn't help either.

mhendriks, 2023-10-18:
make BE items comment is a solution too ;-) But you will run into other dsmr2lib errors because of the GetCRC function which is added in the forked version.

//  /* FixedValue */            ,peak_pwr_last_q
//  /* TimestampedFixedValue */ ,highest_peak_pwr
//  /* String */                ,highest_peak_pwr_13mnd

jeroenrnl, 2023-10-18:
It worked!
My dongle is now running my adapted firmware and I get data in both Home Assistant and my own app (https://gitlab.com/jeroenrnl/nrg - code on Gitlab does not include the needed changes for MQTT as they are still in development).

I will keep it running for a day or so and if I don't run into issue, create a pull request.