mhendriks/P1-Dongel-ESP32 #25: Hostname not accepted as MQTT server

jeroenrnl, 2023-10-14:
When setting the MQTT server to a hostname instead of an IP address, it doesn't work, the telnet interface logs:
```
0 [21:42:29][ 157148|124916] LogFile     ( 377): MQTT Starting
0 [21:42:31][ 158696|124916] connectMQTT_( 139): queryHost MQTT failed
0 [21:42:31][ 155524|124916] connectMQTT_( 147): ERROR: [mqtt.server.com] => is not a valid URL
0 [21:42:31][ 158732|124916] sendMQTTData( 368): no connection with a MQTT broker ..
```
(I replaced the actual name with a fake one)

When setting the IP address, it works

mhendriks, 2023-10-14:
is it an intern hostname. https isn't supported yet...
some times the dongle should be rebooted after a change ...

jeroenrnl, 2023-10-15:
It's an internal hostname (can be resolved using my internal DNS). It's not using https.

Also, I do not see it trying to reach my DNS.

mhendriks, 2023-10-16:
ah clear. Please try again after installing version 4.8.6.
Please, reboot the dongle after changing the broker host

jeroenrnl, 2023-10-16:
Yes, thank you, 4.8.6 fixed that.

closed 2023-10-16 by jeroenrnl