mhendriks/P1-Dongel-ESP32 #30: 4.8.11 has no 3 previous days electricity use

RobvanderDoes, 2023-12-20:
After updating to 4.8.11 the info/circular indications of three days electricity use are empty.

mhendriks, 2023-12-20:
Even after a browser refresh? is the history (days/etc) available?

RobvanderDoes, 2023-12-21:
Yes, even tried several browsers on several devices. All history from before the update has gone.

RobvanderDoes, 2023-12-21:
This includes the water usage history.

RobvanderDoes, 2023-12-21:
[Slimme Meter.pdf](https://github.com/mhendriks/P1-Dongel-ESP32/files/13737412/Slimme.Meter.pdf)
As you can see all data from before the update is missing.

mhendriks, 2023-12-21:
:'-( 
what was the firmware version before updating? Update was tested on version 4.8.5.
I will do a retest on a couple of new dongles

RobvanderDoes, 2023-12-21:
4.8.9, as 4.8.10 wasn't available.

mhendriks, 2023-12-21:
thanks. unfortunately there is a problem as there is also another user with the same problem.
I removed the 4.8.11 update message and the link to the click for update.

On a couple of new dongles I do some updates (via url method and telnet)
4.8.5 -> 4.8.11
4.8.8 -> 4.8.11
4.8.10 -> 4.8.11
4.8.11 -> 4.8.5
4.8.7 -> 4.8.11

I couldn't reproduce the wiping problem:-( 
I think there was a mounting issue after update and wipes the whole data partition. 

On what way did you update the dongle? via the update page/url/click for update on the information page?

RobvanderDoes, 2023-12-22:
I updated via http://p1-dongle-pro.local/remote-update?version=4.8. 11

RobvanderDoes, 2023-12-22:
I updated via http://p1-dongle-pro.local/remote-update?version=4.8. 11
And now yo mentioned it: the frontend.json was back to default.

mhendriks, 2023-12-22:
thanks for the information. That is same update methode I use. 
I will perform more updates to see if it fails on sudden moment.

RobvanderDoes, 2023-12-22:
Extra info, but not sure if this helps: when 4.8.10 was made 'latest' the update didn't work. In the end I updated by downloading the zip-file, and upgraded via the web-interface. But the version info on the web-interface didn't change: it remained on 4.8.9

mhendriks, 2023-12-22:
clear thanks for the additional info. The zip file (from github) will not work. Only .bin files are accepted. The 4.8.10 version is a frontend only version. The binary isn't changed. I think this failed update isn't the reason why the data was whiped. Anyway thanks for the information.

closed 2024-01-07 by mhendriks

RobvanderDoes, 2024-01-07:
Which commit has fixed this issue?

mhendriks, 2024-01-07:
4.8.12 is the latest but the issue couldn't be reproduced :'-(

RobvanderDoes, 2024-01-07:
So at least 2 users experienced this bug but now the report has been closed without being fixed?  After loosing all my data, I don't dare to update any more....

mhendriks, 2024-01-07:
I understand it Rob. I'm not happy with it myself either. The part which could be responsible for the wiping (mounting the flash storage) isn't changed for a while. At this moment it is a quess what it could be. I did much testing updating from old versions and back and forward ... but i didn't ran into this issue.

If this popsup again I will be reopening this issue.

Best practise is to download the RNG and settingfiles before an update.

RobvanderDoes, 2024-01-07:
Would it be wise to add that suggestion to the manual?
Is there a (command line) command that would do it all (download + upgrade + upload)?
Or even better: a GUI command (on the web-interface)?

mhendriks, 2024-01-09:
Thanks for the hint. I add it directly.
There is no command to do this all :-(  Normally this shouldn't happen but it is a good practise to download the files before every update. I will add this feature on the backlog (zip the files, download + update).