mhendriks/P1-Dongel-ESP32 #33: Home assisant auto discovery MQTT topic incorrect

Runesmacher, 2024-01-30:
Hi

The Topic for the water sensor in Home assistant with MQTT seems to be incorrect.
Dongle v4.8.14.

The topic should be changed to water_delivered in the auto discovery

[image]
[image]

mhendriks, 2024-02-04:
@Runesmacher 
I checked it with my current version 4.8.14 b2807 ... looks okay.

[image]

Runesmacher, 2024-02-04:
I have a smart water, gas and electricity meter that all feed trough the Electricity meter, i think that is where the issue stems from.
MQTT send different topics then what HASS is sent as autodiscover:

https://github.com/mhendriks/P1-Dongel-ESP32/blob/7303b3bf7d601df998bc7440a684b0d6187a8c2a/_Water.ino#L8-L17

https://github.com/mhendriks/P1-Dongel-ESP32/blob/7303b3bf7d601df998bc7440a684b0d6187a8c2a/MQTT.ino#L78

mhendriks, 2024-02-04:
Ah yes this is the problem, sloppy error :-( Thanks for noticing me!
It is changed to the water topic in version 4.8.15 ... now available for update :-)

[image]

Runesmacher, 2024-02-04:
Can confirm it is fixed with 4.8.15.

Thanks!

closed 2024-02-04 by mhendriks