mhendriks/P1-Dongel-ESP32 #34: Feature: Posibility to add belgian peak usage data

Runesmacher, 2024-02-05:
Hi

Since i live in flanders, we have peak usage billing, it would be nice if we could also track this.

Reference topic for the default DSMR integration for HASS which provides a lot of info:
https://community.home-assistant.io/t/dsmr-monthly-15-minute-peak-values-for-belgium/493105/37

Runesmacher, 2024-02-15:
I am using DSMR-API indeed, would it be possible to send them to home assisant via MQTT?

mhendriks, 2024-02-15:
I see in the code that the "peak_pwr_last_q" and "highest_peak_pwr" are sent via the mqtt interface.

Runesmacher, 2024-02-15:
Can you then also add it to the auto discovery of Home assistant?

Also did a quick repo search for those values but can;t seem to find it? (I am looking at the master branch)
[image]

Runesmacher, 2024-05-16:
Any update on this?

mhendriks, 2024-05-26:
In my opinion those values are sent via MQTT (and could be read via REST JSON API).
Only omitting issue is the MQTT HA Auto discovery ... but because 99% of the program storage is full modifications like this could be done. The new version (5.0.0) will have some big refactoring improvements and this will gain space.

closed 2025-07-20 by mhendriks