mhendriks/P1-Dongel-ESP32 #35: Feature Request: SSL for MQTT

jeroenst, 2024-02-14:
At this moment there is no option for MQTT over SSL.

Can this be integrated?

mhendriks, 2024-02-14:
correct. It is added to the backlog :-)

jeroenst, 2024-04-30:
Is there any progress on this feature?

mhendriks, 2024-05-01:
no, not yet. The solution is simple but there is not enough program space left to implement the option. I will make a compiler option to enable the sMQTT connection ... the none tsl connection will not work in this version. This is an ugly solution but it is the most feasible at the moment. I will do this within a couple of days.

mhendriks, 2024-05-01:
Jeroen, There are updates. 
P1 Ethernet Pro: http://eth-dongle-pro.local/remote-update?version=4.8.17TLS
P1 Dongle Pro : http://p1-dongle-pro.local/remote-update?version=4.8.17TLS

Note: mqtt is always with TSL ... so unsecure mqtt isn't supported by this version.

jeroenst, 2024-05-01:
Hi!

Thank you, I will test this next week.

jeroenst, 2024-05-22:
I can confirm SSL works. Thank you!

closed 2024-05-22 by jeroenst