mhendriks/P1-Dongel-ESP32 #36: Feature Request: MQTT message on start and end of DSMR telegram

jeroenst, 2024-02-14:
Is it possible to create an MQTT topic for example "dsmr_status" where a message is send upon start and ending of the DSMR telegram?

So on the / character "receiving" is send to the topic dsmr_status
And when the CRC is correct "ready" is send to the topic dsmr_status
And when the CRC has failed "crcfail" is send to the topic dsmr_status

This is mainly needed for Nodered to know when the all data is received and is ready to be written to a database.

Maybe this is also wanted for MBUS?

mhendriks, 2024-02-15:
Mqtt data is only written after a successful p1 telegram has been read. So begin / end and crc checks are passed when publishing

hint: raw telegram is available on poort 82 (after enabling this in the settings).

jeroenst, 2024-02-16:
In my setup, all values are published to different topics so they are received in serial order, not parallel. When the values where published in one message as JSON, there would be no time-difference between the received values.

```
P1-Dongle-Pro/LWT Online
P1-Dongle-Pro/identification XMX5LGBBFG1012700000
P1-Dongle-Pro/mac XX:XX:XX:XX:XX:XX
P1-Dongle-Pro/p1_version 42
P1-Dongle-Pro/equipment_id 4530303330303033373238353233300000
P1-Dongle-Pro/firmware 4.8.14
P1-Dongle-Pro/ip_address 192.168.0.74
P1-Dongle-Pro/wifi_rssi -61
P1-Dongle-Pro/gas_equipment_id 4730303235303033353531313439300000
P1-Dongle-Pro/gas_delivered 8948.510
P1-Dongle-Pro/gas_delivered_timestamp 240216080000W
P1-Dongle-Pro/timestamp 240216082650W
P1-Dongle-Pro/energy_delivered_tariff1 3486.242
P1-Dongle-Pro/energy_delivered_tariff2 8761.247
P1-Dongle-Pro/energy_returned_tariff1 0.566
P1-Dongle-Pro/energy_returned_tariff2 0.985
P1-Dongle-Pro/electricity_tariff 0002
P1-Dongle-Pro/power_delivered 0.110
P1-Dongle-Pro/power_returned 0.000
P1-Dongle-Pro/current_l1 0.000
P1-Dongle-Pro/current_l2 0.000
P1-Dongle-Pro/current_l3 0.000
P1-Dongle-Pro/power_delivered_l1 0.021
P1-Dongle-Pro/power_delivered_l2 0.056
P1-Dongle-Pro/power_delivered_l3 0.033
P1-Dongle-Pro/power_returned_l1 0.000
P1-Dongle-Pro/power_returned_l2 0.000
P1-Dongle-Pro/power_returned_l3 0.000
P1-Dongle-Pro/gas_delivered 8948.510
P1-Dongle-Pro/gas_delivered_timestamp 240216080000W
```

jeroenst, 2024-02-16:
Best option (for me) would be if the bActJsonMQTT can be configured by the web gui so all the data is send in one message.

This also gives less network overhead and traffic.

mhendriks, 2024-02-16:
the hidden bActJsonMQTT (you found it:) option is the way to do this ...
Not standaard because of the legacy (ESPHome) implementations.

I am planning a new version v5 with a lots of improvements. One of them is one proper json payload instead of bursts.
This version will not be a simple update.

jeroenst, 2024-02-16:
I configured "act-json-mqtt":true in DSMRsettings.json. For me this is a good solution.