mhendriks/P1-Dongel-ESP32 #38: Electricity data is not retained

jeroenst, 2024-02-16:
All messages on MQTT are retained except the values for the electricity data. Can this be changed?

Retained Messages:
```
P1-Dongle-Pro/LWT Online
P1-Dongle-Pro/identification XMX5LGBBFG1012765679
P1-Dongle-Pro/mac DC:54:75:BB:D4:8C
P1-Dongle-Pro/p1_version 42
P1-Dongle-Pro/equipment_id 4530303330303033373238353233313137
P1-Dongle-Pro/firmware 4.8.14
P1-Dongle-Pro/ip_address 192.168.0.74
P1-Dongle-Pro/wifi_rssi -61
P1-Dongle-Pro/gas_equipment_id 4730303235303033353531313439333137
P1-Dongle-Pro/gas_delivered 8948.510
P1-Dongle-Pro/gas_delivered_timestamp 240216080000W
```

Not retained messages:
```
P1-Dongle-Pro/timestamp 240216083001W
P1-Dongle-Pro/energy_delivered_tariff1 3486.242
P1-Dongle-Pro/energy_delivered_tariff2 8761.253
P1-Dongle-Pro/energy_returned_tariff1 0.566
P1-Dongle-Pro/energy_returned_tariff2 0.985
P1-Dongle-Pro/electricity_tariff 0002
P1-Dongle-Pro/power_delivered 0.115
P1-Dongle-Pro/power_returned 0.000
P1-Dongle-Pro/current_l1 0.000
P1-Dongle-Pro/current_l2 0.000
P1-Dongle-Pro/current_l3 0.000
P1-Dongle-Pro/power_delivered_l1 0.020
P1-Dongle-Pro/power_delivered_l2 0.059
P1-Dongle-Pro/power_delivered_l3 0.036
P1-Dongle-Pro/power_returned_l1 0.000
P1-Dongle-Pro/power_returned_l2 0.000
P1-Dongle-Pro/power_returned_l3 0.000
P1-Dongle-Pro/gas_delivered 8948.510
P1-Dongle-Pro/gas_delivered_timestamp 240216080000W
```

mhendriks, 2024-02-16:
this is on purpose. The retained ones have a long update interval. The dynamic data a short interval mostly.

jeroenst, 2024-02-16:
Can you make an option for retaining? When I want to create a web frontend on our webserver, I request the data from the MQTT server with an MQTT websocket or PHP script. If the values are not retained, I have to store them manually.