mhendriks/P1-Dongel-ESP32 #41: "act-json-mqtt":true still sends gas values in separate topics

jeroenst, 2024-02-16:
When I enable act-json-mqtt":true gas values are still send into separate topics:

```
getpraut/P1-Dongle-Pro/all {"timestamp":"240216170332W","energy_delivered_tariff1":3486.242,"energy_delivered_tariff2":8765.862,"energy_returned_tariff1":0.566,"energy_returned_tariff2":0.985,"electricity_tariff":"0002","power_delivered":0.124,"power_returned":0.024,"current_l1":0,"current_l2":0,"current_l3":1,"power_delivered_l1":0.068,"power_delivered_l2":0.056,"power_delivered_l3":0,"power_returned_l1":0,"power_returned_l2":0,"power_returned_l3":0.024,"gas":8952.446289,"gas_ts":"240216160000W"}
getpraut/P1-Dongle-Pro/gas_delivered 8952.446
getpraut/P1-Dongle-Pro/gas_delivered_timestamp 240216160000W

```

mhendriks, 2024-02-16:
this shouldn't be.... I changed this in the current code.

jeroenst, 2025-07-20:
Unfortunately the water and gas topics are still send with 4.15.5

mhendriks, 2025-07-20:
These values have been added intentionally.