mhendriks/P1-Dongel-ESP32 #42: Tarief 1,40 wordt opgeslagen als 1.399999976

jeroenst, 2024-02-16:
Als ik een tarief invoer van 1,4 wordt dit opgeslagen als 1.399999976. Waarschijnlijk wordt dit in een float opgeslagen en deze kan de precisie van 1.4 niet aan. Beter is het denk ik om het tarief op te slaan in een uint16_t maal 1000. Dus 1,40 wordt dan 14000 in het geheugen.

Overigens geldt dit eigenlijk ook voor alle andere waarden. kW zou eigenlijk als watt moeten worden opgeslagen, kWh als Wh, m3 als liter en pas op het einde worden voorzien van een punt op de derde positie vanaf rechts.

Beter zou nog zijn op alles uit te sturen in watt, watt uur en liter voor gas, maar ik snap dat het gewenst is dat de data van de slimme meter zonder aanpassingen wordt doorgestuurd.

jeroenst, 2024-02-16:
Sorry I forgot to write in English...

mhendriks, 2024-02-16:
you are right. Better to store all data as integer. Legacy of original open source code ;-) 
This is also nominated to change in a new version. Floating points should be avoided.