mhendriks/P1-Dongel-ESP32 #47: MQTT AutoDiscovery (AD) not working in both HA and Domoticz (DM)

infoedkok, 2024-02-29:
Installed the dongle today and integrated it with HA en DM. 

First with MQTT. Noticed that the items were sent to homeassistant topic but with an onther "flat"structure then I know of other MQTT devices.

E.g , in MQTT Explorer, I see this:

homeassistant>
energy_delivered_tariff1=27727.129

Would expect some other structure for AD? 
But I'm not a MQTT specialist ;-)

So, might that cause the not working AD in HA and DM ?

Switching to the other integration type (DSMR slimme meter by network in HA; P1 smart meter with LAN interface in DM) activates the AD's.

But I would like to get it working with MQTT.

Please advice.

Regards Ed

mhendriks, 2024-03-02:
Hi Ed,

Th flat structure is simple but generates a lot of trafic ... this will be changed in a future versie. V5 of the software.
It is possible to turn the json payload on via a hidden setting in the settings.json file : "act-json-mqtt":false

De AD of the dongle will publish a structure with homeassistant topic (try MQTTExplorer to check this).

Best way to integrate with DM is via de port 82 option. see: https://docs.smart-stuff.nl/smart-home-systemen/domoticz-and-dsmr-reader
But you did already find it ;-)

The AD mqtt publish is once during connecting with the broker.  It is a retained message. Sometimes a reboot or reconnect triggers HA.

Warm regards, Martijn

infoedkok, 2024-03-05:
Hi Martijn,
Had to be patient ;-), after a while the MQTT 's showed up in HA and DM. 
Noticed that "act-json-mqtt":false was already in the setting file. And that indeed creates MQTT / AD topics, e.g
topic: homeassistant/sensor/p1-dongle-pro/gas_delivered/config
{
  "uniq_id": "gas_delivered",
  "dev_cla": "gas",
  "name": "Gas Delivered",
  "stat_t": "homeassistant/gas_delivered",
  "unit_of_meas": "m³",
  "val_tpl": "{{ value | round(3) }}",
  "stat_cla": "total_increasing",
  "dev": {
    "ids": "31682840581340",
    "name": "P1meter",
    "mdl": "P1 Dongle Pro",
    "mf": "Smartstuff"
  }
}

Some strange behaviour in DM (not in HA!): the gas MQTT message causes the creation of a "RFXMeter" -type and "RFXMeter counter" subtype. That type parses data in kWh and not in the "unit_of_meas": "m³" as specified in the MQTT.
I was expecting type "P1SmartMeter" subtype "Gas".

But that's DM-stuff, your dongle works like a charm , thanks a lot!

mhendriks, 2024-03-05:
Hi Ed, good to hear that everything is working well :-) Thanks for your feedback!

closed 2024-03-05 by mhendriks