mhendriks/P1-Dongel-ESP32 #50: Implement wifi connection checking

moedtm, 2024-05-10:
During temporary outage of my wifi network almost always the P1 dongel fails to reconnect once the network is back up again. A simple power cycle is enough to bring it back up, but that means a physical action and presence is needed. 
Is this expected behavior? Could a sort of WiFi watchdog be implemented that reboots if for example WiFi connection is lost?

mhendriks, 2024-05-11:
Which firmware version is installed? Firmware 4.8.15 contains a time out fix for the Wifi network.

moedtm, 2024-05-11:
I updated to 4.8.16 about a week ago, didn't fix the issue unfortunately.

mhendriks, 2024-05-11:
Is it possible to share the log files .log and .old ?

moedtm, 2024-05-11:
Sure! Below the capture of losing the connection.

`{"time":"2024-05-11 16:46:21","log":"WiFi-event : 5 | rssi: 0 | channel : 6"}
{"time":"2024-05-11 16:46:21","log":"Wifi connection lost"}
{"time":"2024-05-11 16:46:21","log":"WiFi-event : 4 | rssi: 0 | channel : 6"}
{"time":"2024-05-11 16:46:21","log":"Connected to . Asking for IP address"}
{"time":"2024-05-11 16:46:21","log":"WiFi-event : 5 | rssi: 0 | channel : 6"}
{"time":"2024-05-11 16:46:21","log":"WiFi-event : 5 | rssi: 0 | channel : 6"}
{"time":"2024-05-11 16:46:21","log":"WiFi-event : 4 | rssi: -56 | channel : 6"}
{"time":"2024-05-11 16:46:21","log":"Connected to . Asking for IP address"}
{"time":"2024-05-11 16:46:21","log":"WiFi-event : 5 | rssi: 0 | channel : 6"}
{"time":"2024-05-11 16:46:21","log":"WiFi-event : 5 | rssi: 0 | channel : 6"}
{"time":"2024-05-11 16:46:21","log":"WiFi-event : 5 | rssi: 0 | channel : 6"}
{"time":"2024-05-11 16:46:21","log":"WiFi-event : 4 | rssi: -56 | channel : 6"}
{"time":"2024-05-11 16:46:21","log":"Connected to D0:21:F9:B3:9B:E5. Asking for IP address"}
{"time":"2024-05-11 16:46:21","log":"WiFi-event : 5 | rssi: 0 | channel : 6"}
{"time":"2024-05-11 16:46:21","log":"WiFi-event : 4 | rssi: 0 | channel : 6"}
{"time":"2024-05-11 16:46:21","log":"Connected to . Asking for IP address"}
{"time":"2024-05-11 16:46:21","log":"WiFi-event : 5 | rssi: 0 | channel : 6"}
{"time":"2024-05-11 16:46:21","log":"WiFi-event : 5 | rssi: 0 | channel : 6"}
{"time":"2024-05-11 16:46:21","log":"WiFi-event : 5 | rssi: 0 | channel : 6"}
{"time":"2024-05-11 16:46:27","log":"MQTT Starting"}
{"time":"2024-05-11 16:46:27","log":"MQTT: Attempting connection... connection FAILED"}
{"time":"2024-05-11 16:46:27","log":"WiFi-event : 5 | rssi: 0 | channel : 1"}`

Continues below, when you see the time change to 16:21 and the reboot is when I plugged the device out and in again. As you see, the device then reconnects within seconds

`{"time":"2024-05-11 16:46:37","log":"MQTT Starting"}
{"time":"2024-05-11 16:46:37","log":"MQTT: Attempting connection... connection FAILED"}
{"time":"2024-05-11 16:46:47","log":"MQTT Starting"}
{"time":"2024-05-11 16:46:47","log":"MQTT: Attempting connection... connection FAILED"}
{"time":"2024-05-11 16:46:57","log":"MQTT Starting"}
{"time":"2024-05-11 16:46:57","log":"MQTT: Attempting connection... connection FAILED"}
{"time":"2024-05-11 16:47:07","log":"MQTT Starting"}
{"time":"2024-05-11 16:47:07","log":"MQTT: Attempting connection... connection FAILED"}
{"time":"2024-05-11 16:47:17","log":"MQTT Starting"}
{"time":"2024-05-11 16:47:17","log":"MQTT: Attempting connection... connection FAILED"}
{"time":"2024-05-11 16:47:27","log":"MQTT Starting"}
{"time":"2024-05-11 16:47:27","log":"MQTT: Attempting connection... connection FAILED"}
{"time":"2024-05-11 16:47:38","log":"MQTT Starting"}
{"time":"2024-05-11 16:47:38","log":"MQTT: Attempting connection... connection FAILED"}
{"time":"2024-05-11 16:47:48","log":"MQTT Starting"}
{"time":"2024-05-11 16:47:48","log":"MQTT: Attempting connection... connection FAILED"}
{"time":"2024-05-11 16:47:58","log":"MQTT Starting"}
{"time":"2024-05-11 16:47:58","log":"MQTT: Attempting connection... connection FAILED"}
{"time":"2024-05-11 16:48:08","log":"MQTT Starting"}
{"time":"2024-05-11 16:48:08","log":"MQTT: Attempting connection... connection FAILED"}
{"time":"2024-05-11 16:48:18","log":"MQTT Starting"}
{"time":"2024-05-11 16:48:18","log":"MQTT: Attempting connection... connection FAILED"}
{"time":"2024-05-11 16:48:21","log":"WiFi-event : 9 | rssi: 0 | channel : 1"}
{"time":"2024-05-11 16:48:28","log":"MQTT Starting"}
{"time":"2024-05-11 16:48:28","log":"MQTT: Attempting connection... connection FAILED"}
{"time":"2024-05-11 16:48:39","log":"MQTT Starting"}
{"time":"2024-05-11 16:48:39","log":"MQTT: Attempting connection... connection FAILED"}
{"time":"2024-05-11 16:48:49","log":"MQTT Starting"}
{"time":"2024-05-11 16:48:49","log":"MQTT: Attempting connection... connection FAILED"}
{"time":"2024-05-11 16:48:59","log":"MQTT Starting"}
{"time":"2024-05-11 16:48:59","log":"MQTT: Attempting connection... connection FAILED"}
{"time":"2024-05-11 16:49:09","log":"MQTT Starting"}
{"time":"2024-05-11 16:49:09","log":"MQTT: Attempting connection... connection FAILED"}
{"time":"2024-05-11 16:49:19","log":"MQTT Starting"}
{"time":"2024-05-11 16:49:19","log":"MQTT: Attempting connection... connection FAILED"}
{"time":"2024-05-11 16:49:29","log":"MQTT Starting"}
{"time":"2024-05-11 16:49:29","log":"MQTT: Attempting connection... connection FAILED"}
{"time":"2024-05-11 16:49:40","log":"MQTT Starting"}
{"time":"2024-05-11 16:49:40","log":"MQTT: Attempting connection... connection FAILED"}
{"time":"2024-05-11 16:49:50","log":"MQTT Starting"}
{"time":"2024-05-11 16:49:50","log":"MQTT: Attempting connection... connection FAILED"}
{"time":"2024-05-11 16:21:47","reboot":"Vbat power on reset","reboots":1035}
{"time":"2024-05-11 16:21:47","log":"Wifi Starting"}
{"time":"2024-05-11 16:21:47","log":"WiFi-event : 0 | rssi: 0 | channel : 0"}
{"time":"2024-05-11 16:21:47","log":"WiFi-event : 2 | rssi: 0 | channel : 1"}
{"time":"2024-05-11 16:21:47","log":"WiFi-event : 4 | rssi: -63 | channel : 6"}
{"time":"2024-05-11 16:21:47","log":"Connected to D0:21:F9:B3:9B:E5. Asking for IP address"}
{"time":"2024-05-11 16:21:47","log":"WiFi-event : 7 | rssi: -62 | channel : 6"}
{"time":"2024-05-11 16:21:47","log":"Wifi Connected"}
{"time":"2024-05-11 16:50:09","log":"MQTT Starting"}
{"time":"2024-05-11 16:50:09","log":"MQTT: Attempting connection... connected"}
`

moedtm, 2024-05-17:
Let me know if I can help testing. A reboot if all else fails still might be the most foolproof solution, even more so since a hard reboot by power cycle immediately solved the problem

mhendriks, 2024-05-17:
Made some changes to the Wifi event manager.
- after 5 min disconnection -> reboot
- proper reconnect handling when router is doing a power cycle within 5 min

the updated 4.8.17 release is available ... let me know if this fix the issue.

moedtm, 2024-05-18:
Nice work, this makes the wifi connection far more reliable. Tried a forced reconnect from the wifi ap a couple of times, instant reconnect or deviation to other ap every time. 
Can be marked as solved! Thanks for fixing this so quickly

mhendriks, 2024-05-18:
:-) great to hear. Thanks for the hint and the logfiles.

closed 2024-05-18 by mhendriks