mhendriks/P1-Dongel-ESP32 #52: WiFi connection unstable

stijnb1234, 2024-05-28:
My dongle has some wifi connection issues. After a day or so, it loses connection and does not reconnect / takes a lot of time to reconnect. It just falls back to Access Point mode without checking if a connection can be made with the same settings again.

I'm not fully sure how to debug / check.

stijnb1234, 2024-05-28:
Logs:
```
{"time":"2024-05-26 20:30:20","reboot":"Vbat power on reset","reboots":391}
{"time":"2024-05-26 20:30:20","log":"Wifi Starting"}
{"time":"2024-05-26 20:30:20","log":"WiFi-event : 0 | rssi: 0 | channel : 0"}
{"time":"2024-05-26 20:30:20","log":"WiFi-event : 2 | rssi: 0 | channel : 1"}
{"time":"2024-05-26 20:30:20","log":"WiFi-event : 4 | rssi: -75 | channel : 6"}
{"time":"2024-05-26 20:30:20","log":"Connected to A2:55:1F:B5:2F:30. Asking for IP address"}
{"time":"2024-05-26 20:30:20","log":"WiFi-event : 7 | rssi: -75 | channel : 6"}
{"time":"2024-05-26 20:30:20","log":"Wifi Connected"}
{"time":"2024-05-26 21:03:25","log":"OTA UPDATE started"}
{"time":"2024-05-26 21:03:25","log":"OTA UPDATE succesfull"}
{"time":"2024-05-26 21:03:25","log":"WiFi-event : 5 | rssi: 0 | channel : 0"}
{"time":"2024-05-26 21:03:25","log":"Wifi connection lost"}
{"time":"2024-05-26 21:03:25","log":"WiFi-event : 3 | rssi: 0 | channel : 0"}
{"time":"2024-05-26 21:03:25","reboot":"Software reset digital core","reboots":392}
{"time":"2024-05-26 21:03:25","log":"Wifi Starting"}
{"time":"2024-05-26 21:03:25","log":"WiFi-event : 0 | rssi: 0 | channel : 0"}
{"time":"2024-05-26 21:03:25","log":"WiFi-event : 2 | rssi: 0 | channel : 1"}
{"time":"2024-05-26 21:03:25","log":"WiFi-event : 4 | rssi: -80 | channel : 6"}
{"time":"2024-05-26 21:03:25","log":"Connected to 62:58:80:2F:78:5F. Asking for IP address"}
{"time":"2024-05-26 21:03:25","log":"WiFi-event : 7 | rssi: -80 | channel : 6"}
{"time":"2024-05-26 21:03:25","log":"Wifi Connected"}
{"time":"2024-05-28 05:33:56","reboot":"Vbat power on reset","reboots":393}
{"time":"2024-05-28 05:33:56","log":"Wifi Starting"}
{"time":"2024-05-28 05:33:56","log":"WiFi-event : 0 | rssi: 0 | channel : 0"}
{"time":"2024-05-28 05:33:56","log":"WiFi-event : 2 | rssi: 0 | channel : 1"}
{"time":"2024-05-28 05:33:56","log":"WiFi-event : 4 | rssi: -76 | channel : 6"}
{"time":"2024-05-28 05:33:56","log":"Connected to A2:55:1F:B5:2F:30. Asking for IP address"}
{"time":"2024-05-28 05:33:56","log":"WiFi-event : 7 | rssi: -76 | channel : 6"}
{"time":"2024-05-28 05:33:56","log":"Wifi Connected"}
{"time":"2024-05-28 17:55:19","log":"WiFi-event : 5 | rssi: 0 | channel : 6"}
{"time":"2024-05-28 17:55:19","log":"Wifi connection lost"}
{"time":"2024-05-28 17:55:30","log":"WiFi-event : 5 | rssi: 0 | channel : 6"}
{"time":"2024-05-28 17:55:33","log":"WiFi-event : 5 | rssi: 0 | channel : 6"}
{"time":"2024-05-28 17:55:43","log":"WiFi-event : 5 | rssi: 0 | channel : 6"}
{"time":"2024-05-28 17:55:45","log":"WiFi-event : 5 | rssi: 0 | channel : 6"}
{"time":"2024-05-28 17:57:19","log":"WiFi-event : 9 | rssi: 0 | channel : 6"}
{"time":"2024-05-28 18:15:10","reboot":"Vbat power on reset","reboots":394}
{"time":"2024-05-28 18:15:10","log":"Wifi Starting"}
{"time":"2024-05-28 18:15:10","log":"WiFi-event : 0 | rssi: 0 | channel : 0"}
{"time":"2024-05-28 18:15:10","log":"WiFi-event : 2 | rssi: 0 | channel : 1"}
{"time":"2024-05-28 18:15:10","log":"WiFi-event : 4 | rssi: -70 | channel : 6"}
{"time":"2024-05-28 18:15:10","log":"Connected to 62:58:80:2F:78:5F. Asking for IP address"}
{"time":"2024-05-28 18:15:10","log":"WiFi-event : 7 | rssi: -72 | channel : 6"}
{"time":"2024-05-28 18:15:10","log":"Wifi Connected"}
```

mhendriks, 2024-05-28:
Hi Stijn, 

Minimal rssi should be -70. At lower rssi you get an unstable wifi connection and packet losses.
The red light should always be on. blinking red light will indicate a power issue = usb charger is needed

closed 2024-06-02 by stijnb1234