mhendriks/P1-Dongel-ESP32 #53: When having two P1 Dongle Pros active at the same time, the HA Auto Discovery option only works for one of them

thijskruithof, 2024-05-30:
I have a Ethernet P1 Dongle Pro running, which monitors my electricity meter and posts its messages to my MQTT server, and has the HA Auto discovery option enabled, so it automatically shows up in Home Assistant as a MQTT device, with all entities automatically configured. 

I now also installed a (very new) Ethernet Warmtelink P1 Dongle Pro, which I'd like to also post its messages to my MQTT server and also automatically show up in Home Assistant. But when I turn on the HA Auto discovery option, Home Assistant immediately loses all entities from the other P1 meter.

I've taken a look at the relevant topics on my MQTT server and it looks like both dongels are posting to the same topics, overwriting each other. The parent topic they're both posting to is named "homeassistant/sensor/p1-dongle-pro". This name also seems to be hardcoded in MQTT.ino (see function SendAutoDiscoverHA) Could that be a bug in the code and should that name perhaps be unique per P1-dongle type? Or maybe even unique per device instance?

For now I've worked around this by only enabling the HA Auto Discovery option on the P1 dongle for my electricity, and for the Warmtelink dongle I've just manually set up some sensors in HA. That works fine too, but is a little bit more manual work.

mhendriks, 2024-05-31:
Thanks for noticing. Nice to hear that the Warmtelink works on the Ethernet version too.

The HA auto detect topic is fixed. It is changed to the DEFAULT HOST NAME in the 4.8.18 version. Just made a new version for the Ethernet Dongle (Smart meter) .... NOT for the Q-Dongle-Pro.

please try this version on the "normal" ethernet dongle.

thijskruithof, 2024-06-01:
Thanks for the very quick reply!

I've updated my eth dongle's firmware using an upload link i found in a previous support ticket, but it still seems to be posting to the same MQTT topic. 

I used this: http://192.168.1.99/remote-update?version=4.8.18

And my p1.log says:
connected"}
{"time":"2024-06-01 09:36:57","log":"OTA UPDATE started"}
{"time":"2024-06-01 09:36:57","log":"OTA UPDATE succesfull"}
{"time":"2024-06-01 09:36:57","reboot":"Software reset digital core","reboots":5}
{"time":"2024-06-01 09:37:27","log":"MQTT Starting"}
{"time":"2024-06-01 09:37:27","log":"MQTT: Attempting connection... connected"}

I also tried manually rebooting it by replugging the p1 cable, but has no effect. I still see new mqtt messages arriving in homeassistant/sensor/p1-dongle-pro. I was expecting a topic named homeassistant/sensor/eth-dongle-pro instead?

Did i maybe use the wrong method for updating the firmware?

My current info:
Huidige Firmware Versie: 4.8.18+2935 (31/05/2024)	
Gecompileerd: May 31 2024 08:19:29
NL of BE: Slimme Meter	NL	
HostName: Eth-Dongle-Pro	

I was previously on an older version, so it definitely installed a new fw version.

mhendriks, 2024-06-01:
yw
The update method is correct. 
I will check the mqtt HA auto discovery later this day

mhendriks, 2024-06-01:
Check done. Made an extra modification ... checked it and does the trick now. 
Please install 4.8.18 again. (is a new build).

thijskruithof, 2024-06-02:
Nice! I installed it on my Eth dongle and it seems to work! It is publishing its auto discovery info to the correct Mqtt topic, a different one from my Q dongle.

I first had to delete all retained messages from my mqtt broker, and then reboot the Eth dongle, and after that the Eth dongle device and entities automatically appear in Home assistant.

So that works nicely, but it looks like there's now a different problem. My Q dongle doesnt appear in Home assistant (have its autodiscovery enabled too and rebooted it too).

It seems to be caused by duplicate values for "uniq_id". My HA log says this:

```
Logger: homeassistant.components.sensor
Source: helpers/entity_platform.py:744
integration: Sensor (documentation, issues)
First occurred: 03:27:58 (42 occurrences)
Last logged: 08:23:54

Platform mqtt does not generate unique IDs. ID current_l1 already exists - ignoring sensor.eth_dongle_pro_current_l1
Platform mqtt does not generate unique IDs. ID current_l2 already exists - ignoring sensor.eth_dongle_pro_current_l2
Platform mqtt does not generate unique IDs. ID current_l3 already exists - ignoring sensor.eth_dongle_pro_current_l3
Platform mqtt does not generate unique IDs. ID gas_delivered already exists - ignoring sensor.eth_dongle_pro_gas_delivered
Platform mqtt does not generate unique IDs. ID water already exists - ignoring sensor.eth_dongle_pro_waterverbruik
```

As a test I've manually published the config of a single sensor, identical to what the Q dongle published, but with its uniq_id value modified. And when i do that the Q dongle device and that single sensor immediately appear in Home Assistant.

I posted this to `homeassistant/sensor/p1-dongle-pro/gas_delivered/config`:

```
{
  "uniq_id": "q_gas_delivered",
  "dev_cla": "gas",
  "name": "Gas Delivered",
  "stat_t": "Q-Dongle-Pro/gas_delivered",
  "unit_of_meas": "m³",
  "val_tpl": "{{ value | round(3) }}",
  "stat_cla": "total_increasing",
  "dev": {
    "ids": "75331995875124",
    "name": "Q-Dongle-Pro",
    "mdl": "P1 Dongle Pro",
    "mf": "Smartstuff"
  }
}
```
Note the `q_` that i added in front of `uniq_id`.

So it seems like each of the sensors should have a more unique id?

Also, it looks like Q dongle doesnt publish a sensor config for the `heat_delivered` sensor? And the sensors it does provide a config for are not being provided by the Q dongle? Could that be a small bug?

thijskruithof, 2024-06-02:
I'm not super familiar with Home Assistant's unique_id usage, but it seems like it should really be unique. I found this: https://developers.home-assistant.io/docs/entity_registry_index/

Which says using a host/device name or IP is not okay. But the device's serial number or MAC address is fine. As long as it is unique and never changes? 

So perhaps uniq_id of each sensor config entry should be something like
<macaddress>_<sensorname> ?

mhendriks, 2024-06-02:
Right this conflicts ...
Only the two values below are published for the warmte link with a q_ added in front.

[image]

this new version for the WarmteLink Ethernet dongle could be installed via  

/remote-update?version=4.8.18ETH

thijskruithof, 2024-06-03:
Nice! It works! I installed that version on my Q dongle and I now see both my Eth dongle and my Q dongle in HA, without any conflicts. And the Q dongle has the correct two sensor entities.

Thanks so much for doing these fixes!

I do see that the timestamp sensor for Q seems to have a small issue. Looks like HA is not getting any value. And I think it's because the discovery config is pointing to Q-Dongle-Pro/heat_ts but that topic doesnt exist. It's correct name seems to be Q-Dongle-Pro/timestamp?

[image]

mhendriks, 2024-06-23:
Just made a 4.8.19ETH version whit some improvements for the MQTT reconnect, SDK update + timestamp renaming.