mhendriks/P1-Dongel-ESP32 #57: P1 Dongel Pro + H2O: No data from smartmeter after update via https://install.smart-stuff.nl/p1p/

SteGe88, 2024-07-22:
After the message from github that there is a new update available via https://install.smart-stuff.nl/p1p/ I have updated my P1 Dongel Pro with H2O meter (I have choosen the right option (DSMR with H2O). Had to choose the erase mode because otherwise it failed.

After the update completed (tried it twice now) there is no message recieved from the smart meter / P1 port. 

It seems I have a lot of new options available but the software showing I have the following firmware: 4.8.18+2918 (18/05/2024)

But I can choose to connect to NRG and Enphase solar power, which wasn't available in the previous firmware.

Is the firmware for the H2O version somehow not correct?

mhendriks, 2024-07-22:
Because of the erase the H2O profile is wiped. Steps to enable this:
1) connect to the dongle with a telnet client
2) after connection press "T4" and enter
3) check of the setting is correct with the command "at" followed by enter; device type should be 4. 
4) reboot the dongle

Everything should be normal.

SteGe88, 2024-07-23:
I just checked and it is giving a 4. I did used the option with H2O in the installer menu. 

Device Type     : 4
Device First Use: 1

mhendriks, 2024-07-23:
Is this a new model hardware (square with screw terminals) or the hexagon shaped one?
The firmware version in the uploader is for the new model. 

T4 = new model
T3 = hexagon model

Change the profile value to T3 if this is a hexagon version.

SteGe88, 2024-07-23:
It was a stupid user mistake…. Didn’t read the device options wrong. Instead of choosing the normal P1 dongle pro I have chosen the H2O+ version which isn’t correct ofcourse. After reflaghing the right version it was working all right!

closed 2024-07-23 by SteGe88

mhendriks, 2024-07-23:
Super, i am glad is fixed :-)