mhendriks/P1-Dongel-ESP32 #58: Version 4.9.1 P1 has timestamp fixed at {"time":"2001-01-01 01:01:01","log":"Wifi Starting"}

HanSjak, 2024-07-24:
After succesfully upgraded to version 4.9.1, RNG files looked fine.
The logging seems to be not OK, time is not refreshedl and 2 reboot lines where not filled out complete.:

P1.log
_,**reboots":1}**
{"time":"2001-01-01   01:01:01","log":"Wifi Starting"}
{"time":"2001-01-01   01:01:01","log":"WiFi-event : 0 \| rssi: 0 \| channel :   0"}
{"time":"2001-01-01   01:01:01","log":"WiFi-event : 2 \| rssi: 0 \| channel :   1"}
{"time":"2001-01-01   01:01:01","log":"WiFi-event : 10 \| rssi: 0 \| channel :   1"}
{"time":"2001-01-01   01:01:01","log":"WiFi-event : 3 \| rssi: 0 \| channel :   1"}
{"time":"2001-01-01   01:01:01","log":"WiFi-event : 11 \| rssi: 0 \| channel :   1"}
{"time":"2001-01-01   01:01:01","log":"WiFi-event : 10 \| rssi: 0 \| channel :   1"}
{"time":"2001-01-01   01:01:01","log":"WiFi-event : 2 \| rssi: 0 \| channel :   1"}
{"time":"2001-01-01   01:01:01","log":"WiFi-event : 1 \| rssi: 0 \| channel :   1"}
**,reboots":2}**
{"time":"2001-01-01   01:01:01","log":"Wifi Starting"}
{"time":"2001-01-01   01:01:01","log":"WiFi-event : 0 \| rssi: 0 \| channel :   0"}
      and several lines "wifi event" more
{"time":"2001-01-01   01:01:01","log":"WiFi-event : 1 \| rssi: 0 \| channel :   1"}
{"time":"2001-01-01   01:01:01","reboot":"Vbat power on   reset","reboots":3}
{"time":"2001-01-01   01:01:01","log":"Wifi Starting"}
{"time":"2001-01-01   01:01:01","log":"WiFi-event : 0 \| rssi: 0 \| channel :   0"}
    and several lines "wifi event" more
{"time":"2001-01-01   01:01:01","log":"WiFi-event : 4 \| rssi: -19 \| channel :   6"}
{"time":"2001-01-01   01:01:01","log":"Connected to 7C:FF:4D:28:09:07. Asking   for IP address"}
{"time":"2001-01-01   01:01:01","log":"WiFi-event : 7 \| rssi: -22 \| channel :   6"}
{"time":"2001-01-01   01:01:01","log":"Wifi Connected"}
{"time":"2001-01-01   01:01:01","log":"WiFi-event : 13 \| rssi: -22 \| channel :   6"}
{"time":"2001-01-01   01:01:01","log":"WiFi-event : 11 \| rssi: -22 \| channel :   6"}
{"time":"2001-01-01   01:01:01","log":"WiFi-event : 10 \| rssi: -22 \| channel :   6"}
{"time":"2001-01-01   01:01:01","log":"WiFi-event : 11 \| rssi: -23 \| channel :   6"}_

mhendriks, 2024-07-24:
The time stamp is obtained form the P1 signal. Is there a P1 signal available?

HanSjak, 2024-07-24:
Yes, as I said, RNG files are updated properly and actual screen is ok.

HanSjak, 2024-07-24:
I just downloaded P1.log and Tim is changes now.
These are the last links of P1.log 
time":"2001-01-01 01:01:01","log":"Connected to 7C:FF:4D:28:09:07. Asking for IP address"}
{"time":"2001-01-01 01:01:01","log":"WiFi-event : 7 | rssi: -22 | channel : 6"}
{"time":"2001-01-01 01:01:01","log":"Wifi Connected"}
{"time":"2001-01-01 01:01:01","log":"WiFi-event : 13 | rssi: -22 | channel : 6"}
{"time":"2001-01-01 01:01:01","log":"WiFi-event : 11 | rssi: -22 | channel : 6"}
{"time":"2001-01-01 01:01:01","log":"WiFi-event : 10 | rssi: -22 | channel : 6"}
{"time":"2001-01-01 01:01:01","log":"WiFi-event : 11 | rssi: -23 | channel : 6"}
{"time":"2024-07-24 11:37:28","log":"WiFi-event : 5 | rssi: 0 | channel : 6"}
{"time":"2024-07-24 11:37:28","log":"Wifi connection lost"}
{"time":"2024-07-24 11:37:28","log":"WiFi-event : 4 | rssi: -19 | channel : 1"}
{"time":"2024-07-24 11:37:28","log":"Connected to 7C:FF:4D:28:09:07. Asking for IP address"}
{"time":"2024-07-24 11:37:28","log":"WiFi-event : 7 | rssi: -23 | channel : 1"}
{"time":"2024-07-24 11:37:28","log":"Wifi Connected"}

HanSjak, 2024-07-24:
To make it more clear: the upgrade to v 4.9.1 finished at prox. 10h30
I checked RNGhours immediately and and also after uploading my old RNGhours 
at 10h45 and at 11h05 again. Everything as expected.

mhendriks, 2024-07-31:
Sounds like everything is okay now?

HanSjak, 2024-08-01:
Its confusing that after upgrading to v4.9.1 before 11h00 and then downloading at 11h02 RNGhours is ok with meterreading 2024072410 and P1.log shows many "Wifi events" and 3 reboots and a logging "Asking for IP address" and "Wifi connected" all with timestamp "2001-01-01 01:01:01", you still get loggings "Wifi event rssi:-22" with timestamp "2001-01-01 01:01:01".
The first plausible timestamp is logged after more then half an hour: {"time":"2024-07-24 11:37:28","log":"WiFi-event : 5 | rssi: 0 | channel : 6"}

One expects start-up timestamps not after the logging of {"time":"2001-01-01 01:01:01","log":"Wifi Connected"}
Logging "Wifi connected" with the then read internet time should be more infomative.
Ideally this is followed by "Connected with P1 succesfully"!

closed 2024-08-01 by HanSjak

HanSjak, 2024-08-01:
It looks if there is a bug that causes a broken "reboot" logging. Not serious, data is written correctly.
I would love the logging of something like "Receiving telegrams starting" with a timestamp from the telegram!

mhendriks, 2024-08-03:
I understand the confusion. 
Good to know:
- the time stamp isn't correct when there isn't a proper P1 reading (available after wifi connection)
- after a while the Dongle status is written to the preferences memory (invisible) 
- after the upgrade to 4.9.x the preference is empty so this startup has no old timestamp
- later startup ... use the latest time stamp from the preferences

Good suggestion that the NTP should be used instead of the P1 time. Logging could be improved :-)

HanSjak, 2024-08-04:
I'm looking forward to this. Thank you!!