mhendriks/P1-Dongel-ESP32 #60: MQTT homeassistant audot discovery not working

githubcdr, 2024-08-20:
Hi,

It seems that the p1 dongle esp32 does not send any updates to the homeassistant/ topic after a reboot. (other messages are arriving)

I enabled the HA discovery flag and reboot my device several times. It should post an update after a reboot if I'm not mistaken?

Kind regards

mhendriks, 2024-08-21:
I think it is because the 4.8.19 / 18 versions doesnt contain the HA auto discovery function due to program size issues. 

The Auto discovery is back in the 4.9.x version. But you should migrate to the 4.9.x via the webinstaller: https://install.smart-stuff.nl/p1p/. 

Beware: all data will be wiped. make a backup of the RNG files before update

githubcdr, 2024-08-21:
Hi, I updated to latest version but there is a bug in the discovery; when you change the topic it won't account the correct topic but will use the default.

closed 2024-08-22 by githubcdr