mhendriks/P1-Dongel-ESP32 #66: documentation to rebuild firmware

theeuwke, 2025-01-10:
I'm trying to rebuild the firmware, but it fails out of the box:
- not clear which libraries should be used, and which versions
- even be installing most of the libraries, some headers are missing (_mqtt_kb.h)
- vague reference errors (mssing macStr in network, MBUS include, while no define).
- direction for creating settings (i've copied hw_profile_p1_pro.h, but no description how to create _secrets/pairing.h)

mhendriks, 2025-01-12:
This master branche only works with SDK 2.0.17. 
SDK 3.x.x will be supported in the 4.10.x+
_mqtt_kb.h is removed. If MQTTKB isn;t used (normally) it will build.

commend #define DEV_PAIRING will solve the _secrets/pairing.h issue. 

The current main version should build.

mhendriks, 2025-01-18:
It is a secret ;-)

I am sorry. Those keys are provided from EnergyID and we can't share those.