mhendriks/P1-Dongel-ESP32 #67: Feature request: real-time water flow

willembargeman, 2025-01-14:
Feature request to add a kind of real-time water flow meter. For example calculate the amount of water usage (liter) last minute and update this every 10 seconds to MQTT.

use case is to trigger actions based on current water usage. Within Home Assistant this is hard the calculate.

mhendriks, 2025-01-15:
Hi Willem,

I understand the need. Did you try https://www.home-assistant.io/integrations/utility_meter/ ?

willembargeman, 2025-01-15:
Yes, I tried Utility Meter. Utility Meter is fine for calculating used energy for last day but not really for real time usage.
Utility Meter will reset every x minutes and is not really a real time meter.

mhendriks, 2025-01-15:
Ah clear. It then needs to be a solution of its own in HA or ESPHome. Unfortunately I have nothing lying around :-(

lcrockett, 2025-05-15:
I would very much like this feature too; real-time water flow. In ESPHome this is achieved using `pulse_meter`, not sure if that can be replicated in the DSMR-API software.

lcrockett, 2025-05-18:
Thank you for the ESPHome example. To use it I would have to swith from DSMR-API to ESPHome. Would it be possible to expose real-time water flow via the DSMR-API ? Cheers

closed 2025-05-20 by mhendriks