mhendriks/P1-Dongel-ESP32 #68: ota.smart-stuff.nl/v5/ seems to be down

jeroenrnl, 2025-02-07:
OTA updates aren't working and when I go to the site in a browser it shows a 403.

Unfortunately, there are no binaries in the Github releases and I don't want to compile them myself :-)

mhendriks, 2025-02-07:
correct it will only work with a deeplink eg. https://ota.smart-stuff.nl/v5/DSMR-API-V4.8.19_4Mb.bin

closed 2025-02-07 by mhendriks

jeroenrnl, 2025-02-07:
```

/!\ UPDATE MODULE /!\
Enter firmware version (eg. 4.5.1): 4.10.3

!!! OTA UPDATE !!!
OTA versie: 4.10.3 | flashsize: 4 Mb
OTA path: http://ota.smart-stuff.nl/v5/DSMR-API-V4.10.3_4Mb.bin
OTA ERROR: (-102): File Not Found (404)
```

mhendriks, 2025-02-07:
up from 4.9.x the url has been changed to https://ota.smart-stuff.nl/p1p/DSMR-API-V4.10.2_4Mb.bin

Note: upgrade from 4.8.x to 4.9.x+ is only possible via the webinstaller due to the partitioning. 

installer: https://install.smart-stuff.nl/p1p/