mhendriks/P1-Dongel-ESP32 #69: Two SolarEdge inverters

Mike19700, 2025-03-27:
The extra info for a SolarEdge inverter is a good one but... I have 2 inverters.
Is it possible to add another one?

mhendriks, 2025-03-29:
The dongle will support only one token and siteID and obtains the daily information form the SolarEdge portal. If this is the total of both it will work other wise ...

RobvanderDoes, 2025-03-29:
Does this apply to all supported inverters? Or is it/would it be possible to have direct access to them, not via a web portal?

mhendriks, 2025-03-29:
Enphase is a local IQ Gateway connection. 
Unfortunately the SolarEdge API is only available via the cloud :-(

mhendriks, 2025-03-30:
Ah, clear. I mean to say that the SolarEdge data cannot be read locally via a network/ip connection (as far as I know). RS485 would indeed be possible but I haven't looked into that yet ;-)

Mike19700, 2025-04-01:
Sorry. I'm straying from the topic.
Perhaps I have not been entirely clear with my question.
Is there any way to be able to add a 2nd inverter through, say, a check mark and a few extra input fields? Then all you basically have to do is add up the values and show them as a total.