mhendriks/P1-Dongel-ESP32 #70: Feature Request: Update via MQTT topic (and Home Assistant)

ExquisiteMetropolis, 2025-04-07:
I see regular updates in the repo, appreciate the improvements to the product.
What I would love to see is the ability to kick-off the update via a MQTT topic (and the ability to issue that from Home Assistant).
(I use the DSMR firmware variation).

Currently the update process is manual work by browsing to the dongle, navigate to the system page and click the update link.
Which basically issues the update via URL (http://<IP-dongle>/remote-update?version=<number>

I'd prefer to see a notification in Home Assistant, that (preferably) also displays the changelog (just like update for an addon).
The notification includes an update button that basically kicks off the OTA update via MQTT on the dongle.

This idea can be compared to how the firmware update process of a Hue Lightbulb works via Zigbee2MQTT.
Z2M detects the update, presents a notification in HA, this is acknowledged by the end-user, HA sends a MQTT topic update command which Z2M picks up and initiates the OTA. 
(Ref: https://www.zigbee2mqtt.io/guide/usage/ota_updates.html#automatic-checking-for-available-updates)

mhendriks, 2025-04-07:
publish with dongle_topic/update with payload 4.12.5 will start the update.
on the /firmware topic the version number is available.

mhendriks, 2025-04-07:
Not all the topics are shared as Auto Discovery Elements. The update and vital data isn't shared. But all the data could be shown in a tool like MQTTExplorer.

mhendriks, 2025-04-12:
Clear, nice feature to implement. It will be on the backlog for now.

awraeven, 2026-08-28:
I've installed 5.8.13 (manually, as I was on the 5.9.0 alpha still).
Is it correct that the new(er) beta version won't show up as HA alert for updates?
I'm eagerly awaiting the official bump to a higher version to see the magic in HA. :-)

mhendriks, 2026-08-28:
Correct, only stable version updates are shown in HA. 
Or ... install 5.8.11 (via <dongle url>/remote-update?version=5.8.11)  and try again ;-)

awraeven, 2026-08-28:
Done, am I missing something? 
In HA I see:

[image]

mhendriks, 2026-08-29:
On a Ethernet dongle pro.

[image]

Check: is the new stable version visable in the p1 dongle pro system info page?

awraeven, 2026-08-29:
In the dongle I see:
[image]

HA shows:
[image]

NOTE: Screenshot shows 37 seconds, this was after I reloaded MQTT Services within HA.
That didn't change/detect the new version.
Before reloading MQTT it showed '16 hours ago' it seems there is an interval on this version via MQTT? 

After rebooting the dongle the new version was visible in HA:

[image]

And updating worked from HA. 
Question remains, is there a detection interval on this value?

mhendriks, 2026-08-29:
The dongle web interface checks the version manifest directly, while Home Assistant only sees the last retained update state published via MQTT. Reloading the MQTT integration in HA does not trigger a new manifest check on the dongle. 

Until now, the dongle checked for a new version at startup and once per night between 01:00 and 06:00. That explains why the new version became visible immediately after rebooting the dongle.

I have prepared the following improvements:

- Check the manifest when MQTT successfully connects or reconnects.
- Continue checking once per night while MQTT or automatic updates are enabled.
- Do not perform daily checks when both MQTT and automatic updates are disabled.
- Retry a failed manifest check after 30 minutes instead of treating that day’s check as completed.

The automatic update setting only controls whether an available update is installed automatically; it no longer affects whether the version is reported to Home Assistant.

This will be included in version 5.9.3

awraeven, 2026-08-29:
Sounds good! Thanks for the quick revision. 👍

awraeven, 2026-09-28:
Something still off?
I see last week 5.9.5 was released.
My unit still is on 5.8.13 and no update notification in HA was received.

awraeven, 2026-09-30:
Additional info, maybe something wrong with providing the payload to HA?
Doesn't seem to update?
HA Rebooted 2 days ago, dongle 20 days ago. 

[image]

[image]

[image]

awraeven, 2026-10-02:
Showed up just 7h ago in HA.
Can't seem to find out why it took so long. What is the interval?

mhendriks, 2026-10-04:
Your dongle was running 5.8.13, so it does not yet include the improvements introduced in 5.9.3.
In 5.8.13, firmware checks run at startup and once per night between 01:00 and 06:00. If a check fails, it waits until the next night. Your screenshots show that HA received latest_version: 5.8.13, which explains why it displayed “up-to-date”.
From 5.9.3, failed checks are retried after 30 minutes, and MQTT reconnects also trigger a check. This should be ok with the next stable release update.

closed 2026-10-06 by awraeven