mhendriks/P1-Dongel-ESP32 #72: Flashing does not work

Edzelf, 2025-04-16:
I want to re-flash my P1-dongle-pro with the on-line esptool.
Connection toe the ESP32 is okay, but the I got the error:
`Downlading firmware p1-dongle-pro-esp32/p1-dongle-pro-esp32.bin failed: 404`
Could it be that the file has disappeared?

Greetings,
Ed.

mhendriks, 2025-04-16:
Hi Ed,

For the P1 Dongle Pro this installer should be used: https://install.smart-stuff.nl/p1p/
Which option did you select?

Edzelf, 2025-04-17:
Hallo Martijn,
Thank you for your answer.  I tried [this link](https://mhendriks.github.io/esphome-p1/).  The filename in the manifest does not exist.
Anyway, the version on smart-stuff.nl worked and my dongle is repaired!

closed 2025-04-17 by Edzelf