mhendriks/P1-Dongel-ESP32 #75: Feature Request: Uptime via MQTT (to HomeAssistant Sensor)

ExquisiteMetropolis, 2025-06-15:
Would it be possible to add the Uptime value in the dongle, as sensor to Home Assistant (via MQTT)?

[image]

I'm not seeing it in the MQTT topics:

[image]

mhendriks, 2025-06-15:
Sure, this is possible. I put it on the backlog.

ExquisiteMetropolis, 2025-07-24:
I see it since https://github.com/mhendriks/P1-Dongel-ESP32/releases/tag/4.15.6 in my MQTT explorer. 
It's however not showing in Home  Assisting in MQTT Info or as a sensor.
Is there a 2nd step needed? 

I didn't have to setup any of the other sensors manually either, so I'm guessing some config piece is missing somewhere?

[image]

[image]

mhendriks, 2025-10-08:
It is part of the static data ... It updates once every 30 min :-) 
It could be moved to the dynamic part so it updates on every interval

mhendriks, 2025-10-19:
Thanks, fixed in the 4.16.1 Beta version ... will be released today

closed 2025-11-13 by awraeven