mhendriks/P1-Dongel-ESP32 #77: gas_delivered and water not being sent via MQTT

jeroenrnl, 2025-06-27:
I recently upgraded my P1-dongle-pro from 4.8.x to 4.12.x and have lost gas and water statistics in home assistant.

I have updated to the latest version available via the OTA update (4.13.6) today.
The topics `gas_delivered` and `water` are not being sent. The web interface shows them so they are coming in to the P1 Dongle. Also, the `all` topic contains the gas usage, but not the water usage. However, the `homeassistant/sensor/p1-dongle-pro/gas_delivered/config` and `homeassistant/sensor/p1-dongle-pro/water/config` topic are still announcing availabilty of the `gas_delivered` and `water` topics.

I have looked at the code and it seems that this is the problem:
```
  if ( bActJsonMQTT ) {
    String buffer;
    if ( gasDelivered ) {
      jsonDoc["gas"] = gasDelivered;
      jsonDoc["gas_ts"] = gasDeliveredTimestamp;
    }
    serializeJson(jsonDoc,buffer);
    MQTTSend("all", buffer, false);
  } else {
	  if ( DSMRdata.highest_peak_pwr_present ) MQTTSend( "highest_peak_pwr_ts", String(DSMRdata.highest_peak_pwr.timestamp), true);

	  MQTTsendGas();
	  MQTTsendWater();
  }  
```

When `bActJsonMQTT` is active, it is not sending gas and water, however, it should do that any way when home assistant auto discovery is on. This is almost exactly the same issue as Issue #24.

I changed the `} else {` line into

```
  }

  if ( !bActJsonMQTT || EnableHAdiscovery) {
```
But I'm unable to compile my changes and test due to #66

mhendriks, 2025-06-28:
Thanks for raising this issue. 
The code is changed to the code below and available in de 4.14.3 version. The fields are always filled. If unavailable those values are 0.

```  if ( bActJsonMQTT ) {
    String buffer;
      jsonDoc["water"] = waterDelivered;
      jsonDoc["water_ts"] = waterDeliveredTimestamp;
      jsonDoc["gas"] = gasDelivered;
      jsonDoc["gas_ts"] = gasDeliveredTimestamp;
    serializeJson(jsonDoc,buffer);
    MQTTSend("all", buffer, false);
  } else {
	  if ( DSMRdata.highest_peak_pwr_present ) MQTTSend( "highest_peak_pwr_ts", String(DSMRdata.highest_peak_pwr.timestamp), true);
	  MQTTsendGas();
	  MQTTsendWater();
  } 
```

jeroenrnl, 2025-07-11:
I have manually upgraded to 4.14.4 (using `remote-update?version=4.14.4`) but I'm not getting any updates on the `water` or `gas_delivered` topics.

mhendriks, 2025-07-13:
With "act-json-mqtt":true enabled in the settings, all data appears in the "all topic" section. I also see water and gas listed.

[image]

Without "act-json-mqtt":false the data is also listed. In my case only gas (no water sensor attached)

[image]

The HA auto discovery only works when "act-json-mqtt":false

jeroenrnl, 2025-07-13:
The whole point of this bug was to make it work with both `act-json-mqtt` *and* HA Autodiscovery. This worked before and is broken since I upgraded from 4.8.x

mhendriks, 2025-07-14:
ah clear. changed it conform 4.8.19 version. It will be available in the 4.15.5 release.

jeroenrnl, 2025-07-16:
I have just updated and see water coming in again, thanks. Gas not yet, but maybe I need to wait until the next hour. I will check again tomorrow morning.