mhendriks/P1-Dongel-ESP32 #78: P1-Dongle-Pro loses connection on Wifi AP reboot, does not reconnect.

ExquisiteMetropolis, 2025-07-09:
I have a few Unifi Access Points in my network.
When I do a change that requires a reboot of the AP's (update for example) the P1 Pro dongle (obviously) loses connection.
Where all other devices in my environment reconnect once the AP has rebooted, the P1 Pro dongle, does not. It stays disconnected.
It remains powered by the P1 meter, however only a physical reconnection (power cycle) of the P1 cable forces the device to reconnect (maybe the reset button too, cable is easier). 

I waited 2 hours for the unit to automatically attempt to re-connect to my wifi. It didn't so forced a power cycle.

This is totally reproducible as this isn't the first time I've noticed this.

Do you need anything from my end to troubleshoot this issue?
I'm running 4.14.2 (DSMR)

ExquisiteMetropolis, 2025-07-16:
Please find the logs below. 

[P1.log](https://github.com/user-attachments/files/21247618/P1.log)

[P1_logold.log](https://github.com/user-attachments/files/21247619/P1_logold.log)

mhendriks, 2025-07-18:
thanks again for sharing. All the Software reset CPU logs are due to timeouts or updates.  As designed.

The "reboot":"Software reset digital core","reboots":3541 rows are not okay. 

What is the mqtt publishing interval?

ExquisiteMetropolis, 2025-07-18:
The publishing interval is set to 10 seconds.
Other MQTT devices have no issue.

mhendriks, 2025-07-19:
clear, thanks. 
The dongle should also handle the MQTT messages correctly without a reboot.

Unfortunately, I can't reproduce the problem. I've turned the Wi-Fi connection off and on several times and also restarted the broker occasionally. I suspect the boot is coming from one of the libraries or the SDK. Perhaps triggered by certain Wi-Fi events.

I've added more logging to version 4.15.6. I've updated the mqtt client library and added a check at one point.

Can you see if this version provides an improvement or more insight into what is causing the issue?

Next step is to update the SDK.

ExquisiteMetropolis, 2025-07-19:
Thanks! I have updated now to 4.15.6, I will post the log once a next reboot occurs. 

Could it also be dongle hardware related?

ExquisiteMetropolis, 2025-07-20:
[

[P1.log.txt](https://github.com/user-attachments/files/21333600/P1.log.txt)

](url)

Should I update? I have the following log.

ExquisiteMetropolis, 2025-07-20:
I've installed 15.6 
The voltage issue, i've had that before (early 2024). In the end Liander replaced the entire P1 meter.
I doubt they will do this again...
Unfortunately, I do not have to option to connect a USB adapter as there is no power socket nearby so I am 'stuck' with power via DSMR. 

I'm not noticing the reboots, the issue was that upon a config change from unifi (the ap's apply there new config/reboot), the P1 meter unit didn't want to reconnect until a manual power cycle (cable out/in). 
I'm not sure if the 15.6 is intended to resolve that issue, I'll monitor. 

Thanks for the help thusfar! appreciated.

ExquisiteMetropolis, 2025-07-29:
Not sure if related, i noticed a different type of reboot today:

`
{"time":"2025-07-29 22:17:20","log":"reboot: Wifi failed to connect and hit timeout"}
{"time":"2025-07-29 22:17:20","reboot":"12 - Reset by JTAG/Button","reboots":3605}
{"time":"2025-07-29 22:17:20","log":"Wifi Starting"}
{"time":"2025-07-29 22:17:20","log":"Network-event : 101 | rssi: 0 | channel : 0"}
{"time":"2025-07-29 22:17:20","log":"Network-event : 110 | rssi: 0 | channel : 1"}
{"time":"2025-07-29 22:17:20","log":"Connected to 18:E8:29:14:D0:E7. Asking for IP address"}
{"time":"2025-07-29 22:17:20","log":"Wifi Connected"}
{"time":"2025-07-29 22:24:08","log":"MQTT: DISCONNECTED to broker ... try to reconnect"}
{"time":"2025-07-29 22:24:08","log":"MQTT: CONNECTED to broker"}
{"time":"2025-07-29 22:23:58","reboot":"15 - Reset due to CPU lock up (double exception)","reboots":3606}
{"time":"2025-07-29 22:23:58","log":"Wifi Starting"}
{"time":"2025-07-29 22:23:58","log":"Network-event : 101 | rssi: 0 | channel : 0"}
{"time":"2025-07-29 22:23:58","log":"Network-event : 110 | rssi: 0 | channel : 1"}
{"time":"2025-07-29 22:23:58","log":"Connected to E0:63:DA:65:FC:26. Asking for IP address"}
{"time":"2025-07-29 22:23:58","log":"Wifi Connected"}
{"time":"2025-07-29 22:24:19","log":"MQTT: DISCONNECTED to broker ... try to reconnect"}
{"time":"2025-07-29 22:24:19","log":"MQTT: CONNECTED to broker"}
`

mhendriks, 2025-09-13:
There are some issues with reconnecting to the wifi network. After a reboot / maintenance run of the network the dongle woudn't reconnect. It should do because auto reconnect is enabled.

I couldnt reproduce the issue with a single wifi modem ( tested rebooting and switching channels). Maybe it is related to mesh networks but .... there is an issue with wifi reconnecting in the 3.3.0 sdk. 

I made a 4.15.14 beta release for the Pro dongle based on the previous sdk (3.2.1) perhaps this will solve the issue ...

mhendriks, 2025-09-21:
Thanks again for the logging. 
I think I found it... this one was hard.

The Wifi connection is restored correctly the first time the connection is lost. The second time, it fails; the Wi-Fi is automatically turned off, and the dongle goes into an infinite wait state. This looks like a bug in the SDK.
I just tested a workaround. An new 4.15.14 Beta release with today's date.

mhendriks, 2025-09-21:
Absolutely true. If the reproduction is so simple afterward, it doesn't feel great. But ultimately, it does give you a satisfied feeling to have found a solution... hopefully ;)

Thanks for your cooperation and perseverance. 

I'm very curious about your findings with these adjustments :)

daxy01, 2025-09-22:
I'm running into the same as well. When I reboot my AP, the P1 dongle doesn't connect to another AP and won't reconnect to the AP when it's back. I'm on 4.15.14 already. Can I install the beta via OTA?

One observation: I'm, also on Unifi, perhaps that is the culprit here?

mhendriks, 2025-09-22:
Please try the 4.15.14 of yesterday (reinstall). The could be done via OTA. Note; the 4.15.14 with the latest bugfix is only available for the P1 Dongle Pro (Not the +).

mhendriks, 2025-09-22:
thanks for testing and feedback. Great to know this solves the bug !!!

daxy01, 2025-09-24:
I'm afraid I had a WiFi disconnect again last night requiring a reboot. The only interesting/relevant log message is:
{"up":103209,"time":"2025-09-23 16:30:37","Wifi connection lost - DISCONNECTED | reason: 34"}

I think I'll be trying out esphome to see if that remains stable. I have a P1-Dongle-Pro, the hexagonal one with water meter. Although it's not related to this issue, may I ask to confirm the pins used? I believe status_led is on pin 7, UART rx_pin 10 for P1 data and water mbus_id 2?

mhendriks, 2025-09-28:
Thanks daxy01 for the feedback. 

The disconnect routine has been adjusted to check which reasons are not automatically handled in SDK 3.3.x. If this occurs, the dongle tries to  reconnect. 

A new beta version 4.15.14 is now available (Pro only).

awraeven, 2025-10-18:
Running 4.16.0 for a while now (OCT 5 2025, 22:26:56), had a power failure (outside cause) updated my network core switch (firmware update) and still the dongle is up & running. Looks to me the issue is to be closed. 

Thanks @mhendriks for the help, patience and support! :)

closed 2025-10-18 by awraeven